paypal/gatt #15: sample.go crashing

hmatulis, 2015-01-23:
Hi, I just installed go and gatt. 
After running: sudo /usr/local/go/bin/go run sample.go 
I'm getting the error message:
panic: runtime error: invalid memory address or nil pointer dereference
[signal 0xb code=0x1 addr=0x0 pc=0x88648]

goroutine 1 [running]:
github.com/paypal/gatt.(*Server).start(0x10426090, 0x0, 0x0)
        /usr/local/go/src/github.com/paypal/gatt/server_linux.go:74 +0xa0
github.com/paypal/gatt.(*Server).AdvertiseAndServe(0x10426090, 0x0, 0x0)
        /usr/local/go/src/github.com/paypal/gatt/server.go:90 +0xe0
main.main()
        /home/pi/google_drive/gatt/sample.go:56 +0x4d0
exit status 2

Any hint on how to proceed?
HCI is down and also bluetooth service

thanks
Henrique

roylee17, 2015-01-23:
Hi @hmatulis, could you show the output of "hciconfig", and "uname -a"?
It requires the kernel version v3.14 or above. (Or you can refer to the doc.go to apply relevant patches)
Also, try "sudo hcitool -i hci0 lescan" to make check your device supports BLE as well.

hmatulis, 2015-01-24:
Hi @roylee17 ,  thanks!!!
I just bought a new Pi and I wrongly assumed the kernel was up to date. but it has the 3.12.35 one instead.
I upgraded the kernel to 3.18.3 and it worked like a charm.
I'm using a CSR 4.0 dongle and it was already working with Bleno.

closed 2015-01-24 by hmatulis