paypal/gatt #16: Unhandled LE LTK Request while connection to IOS

hmatulis, 2015-01-25:
Hi ,
I'm using the "sample.go"  on a RPI 3.18 
I can connect to mac / os and inquire services / characteristics and exchange data. 
But while connecting to very simple IOS app , the IOS app sends an LTK request and the sample.go shows on screen : "hci: unhandled LE event: LE LTK Request" 
on the IOS side, the app is exiting with an  "Error Domain=CBErrorDomain Code=6 "The connection has timed out unexpectedly." UserInfo=0x174072280"
I tried also with a few apps ( LightBlue, BLE diag tool) and all of them are giving the same result. On IOS I couldn't found an alternate method of connecting.
Thanks
Henrique

roylee17, 2015-01-25:
Currently, we use HCI_CHANNEL_USER and do the l2cap stuff on our own. And the LTK is not supported yet, though it should not take too much effort as we've got the building blocks.

I've been using LightBlue across iPhone 5s/6, iPod touch for testing and diag as well. It works well  without any LTK stuff. However, iOS tends to cache the device info as possible as it can. So, I suspect, the previous failure with the simple app might cause side effect to the subsequent attempts with other apps. You can cleanup the cache by killing all apps that access to bluetooth and then turn the bluetooth off and on before starting next app.

By the way the following commands can dump the HCI traffics.

   sudo hcidump -i hci0 -x -t

You can collect the following cases for comparison and reference
 * Mac to go-gatt on RPI
 * iOS app to go-gatt on RPI
 * iOS app to bleno/bluez/any other gatt server implemetnatation that works on RPI.

Thanks.
Roy

hmatulis, 2015-01-26:
Hi,
Thanks again for your help!
I tried with another iPhone and it worked...
My iPad cached the info in a way I couldn't reset it. I tried turning it on and off, turning bluetooth on/off, turning wi-fi on/off, uninstalling the app, making a clear build.
But I ended with this workaround:
I compiled btaddr tool of bluez package and changed the virtual mac address of the bluetooth device:

sudo hciconfig hci0 up (must be working to change the address)
sudo bdaddr -i xx:xx:xx:xx:xx  (xx:xx:xx:xx:xx:xx new address)  - note I just incremented the previous address by one. 
sudo hciconfig hci0 down (to test with gatt in go) 
Thanks!

closed 2015-01-26 by hmatulis

josharian, 2015-02-02:
@roylee17 do you think it is worth automatically bumping the bdaddr on server startup? Or at least provide a hook in the API to do so? iOS's caching behavior is very frustrating, and having `gatt` just fix it during development would be pretty awesome.

roylee17, 2015-02-02:
@josharian Sure. That did bite me before I realize it. So I've been manually changing my MAC as well.

I was thinking to add an Option for specifying the MAC, and set it nil would randomly generates one. I'll work on it when I get the 'next' branch more stable, and merge compatible changes to the master.

josharian, 2015-02-02:
@roylee17 SGTM