paypal/gatt #17: Native linux bluetooth device enumeration and info

wolfeidau, 2015-01-27:
Just wanted to sync up with the other work going on here.

Over the weekend I completed a basic replacement for the bluez methods:

* `hci_for_each_dev`
* `hci_devinfo`

Just wanted to know if there was already work happening in this area? And whether I should make a pull request so you can decide whether you want this functionality.

I am working on integrating mainly client functions at the moment.

cc @roylee17

roylee17, 2015-01-27:
Hi @wolfeidau, 

That's great, please send a PR so we can merge the effort.

I've been also working on the client support and Mac support.
Some code/naming/data structures changes for this purpose, and results large LOC change.
So we create a "next" branch to move forward without breaking current master branch.
I made some progress, and currently the client functions are in good shape for both Linux and Mac.
Though the API are still subject to change before the server support for Mac also working (should be soon). I'll try cleanup and get the initial implementation to the "next" branch tomorrow.

For API compatible changes, such as the proposed replacement of hci_for_each_dev, hci_devinfo, we can merge them in the master branch. I'll also merge them to the next branch as possible. For more large changes, such as the client support, we'll how see how to integrate as we can. Or if the next branch doesn't divert from your expectation, we can joint our effort to shape it better :-)

roylee17, 2015-01-27:
Hi @wolfeidau,

I pushed a snapshot on at: https://github.com/roylee17/gatt/tree/next-v1
It's rough, but I'll try hard to get more details done this week before merging it to the next branch here.
And then will mark more inline TODO/FIXME, and create issues here.

We'll keep improve both master and next branches. So if you prefer to stay with master branch, I'll still port your changes to the 'next' branch if possible.

```
# scan for peripheral id, MAC address for Linux, UUID for OSX
go run examples/discover.go

# explorer specified 
go run examples/explorer.go [peripheral-id]
```

wolfeidau, 2015-01-27:
Just pushed up the code I pulled together over the weekend to enumerate bluetooth devices in linux. It is committed at https://github.com/wolfeidau/gatt/tree/device_enumeration.

I have a separate module for `ioctl` macro replacements, this is probably something you already have in the branch you just pushed. 

I tried to keep the code relatively self contained while I worked out how to do the ioctl calls. More than happy to massage it however you like to get it merged in with your work.

roylee17, 2015-01-27:
Nice work, we can't miss this. I'll merge it when I integrate it with the other staging code.

Thanks a lot.

wolfeidau, 2015-01-27:
Great stuff, I will do some review of your branch tonight. Looks like lots of new stuff to try out!

roylee17, 2015-03-09:
Integrated in the current master branch.
Thanks.

closed 2015-03-09 by roylee17