paypal/gatt #29: Error while running examples/server.go

wujiang, 2015-03-30:
- Platform

  ```
Darwin Neptune 14.1.0 Darwin Kernel Version 14.1.0: Mon Dec 22 23:10:38 PST 2014
  ```

- Repro
  - run `server.go`: `go run examples/server.go`
  - select the peripheral on LightBlue

- stack trace

  ```
~GOPATH/src/github.com/paypal/gatt $ go run examples/server.go
State: PoweredOn
panic: runtime error: invalid memory address or nil pointer dereference
[signal 0xb code=0x1 addr=0x30 pc=0x405828f]

goroutine 17 [running, locked to thread]:
github.com/paypal/gatt.(*central).stopNotify(0x0, 0xc208030180)
        /Users/wjiang/workspace/go/src/github.com/paypal/gatt/central_darwin.go:64 +0x1f
github.com/paypal/gatt.(*device).respondToRequest(0xc20805a000, 0x16, 0xc20803a9c0)
        /Users/wjiang/workspace/go/src/github.com/paypal/gatt/device_darwin.go:312 +0x23a6
github.com/paypal/gatt.(*device).HandleXpcEvent(0xc20805a000, 0xc20803a990, 0x0, 0x0)
        /Users/wjiang/workspace/go/src/github.com/paypal/gatt/device_darwin.go:348 +0x489
github.com/paypal/gatt/xpc.handleXpcEvent(0x7200140, 0x4304e10)
        /Users/wjiang/workspace/go/src/github.com/paypal/gatt/xpc/xpc_darwin.go:199 +0x28b

goroutine 1 [select (no cases)]:
main.main()
        /Users/wjiang/workspace/go/src/github.com/paypal/gatt/examples/server.go:51 +0x24e

goroutine 5 [chan receive]:
github.com/paypal/gatt.(*device).loop(0xc20805a000)
        /Users/wjiang/workspace/go/src/github.com/paypal/gatt/device_darwin.go:448 +0x83
created by github.com/paypal/gatt.(*device).Init
        /Users/wjiang/workspace/go/src/github.com/paypal/gatt/device_darwin.go:54 +0x58

goroutine 18 [syscall, locked to thread]:
runtime.goexit()
        /usr/local/Cellar/go/1.4/libexec/src/runtime/asm_amd64.s:2232 +0x1
exit status 2
  ```

Please let me know if you need more information. Thank you.

roylee17, 2015-04-02:
@wujiang Thanks for reporting, could you help to confirm the fix works for you as well?

wujiang, 2015-04-03:
@roylee17, thank you for responding so fast.

The original problem is gone. But when I used LightBlue to interrogate it, I got this error:
[image]

It looks like the `d.subscribers` in line https://github.com/paypal/gatt/blob/a1218a6773043bc92bb2581a9686e03ac373017f/device_darwin.go#L311 is an empty map. Not sure if this is an issue.

roylee17, 2015-10-11:
haven't seen it for a long time.
I'm closing this issue now, but feel free to reopen it if any one see this again.

closed 2015-10-11 by roylee17