paypal/gatt #40: scan doesn't work on Intel Edison

shirou, 2015-06-03:
From https://github.com/paypal/gatt/pull/20, I expected gatt works on Intel Edison, but not worked.

```
root@edison:~/src/github.com/paypal/gatt/examples# go run discoverer.go 
2015/06/03 04:14:11 dev: hci0 up
2015/06/03 04:14:11 dev: hci0 down
2015/06/03 04:14:11 dev: hci0 can't bind to hci user channel, err: invalid argument.
2015/06/03 04:14:11 dev: hci0 opened
State: PoweredOn
scanning...
```

examples/discover.go starts scanning but nothing happen.

My environment is below. To avoid cross compiling issue, I have started from self build on Intel Edison.

```
- go version go1.4.2 linux/386
- Linux edison 3.10.17-poky-edison+ #1 SMP PREEMPT Wed Apr 29 03:54:01 CEST 2015 i686 GNU/Linux
- image: edison-image-ww18-15
```

Thanks!

roylee17, 2015-06-03:
We rely on kernel v3.14+ though you can find the patches for older kernels.
v3.10 should have no conflict apply those patches if I remember correctly :-)
 
``` 
To gain complete and exclusive control of the HCI device, gatt uses HCI_CHANNEL_USER (introduced in Linux v3.14) instead of HCI_CHANNEL_RAW. Those who must use an older kernel may patch in these relevant commits from Marcel Holtmann:

Bluetooth: Introduce new HCI socket channel for user operation
Bluetooth: Introduce user channel flag for HCI devices
Bluetooth: Refactor raw socket filter into more readable code
```

shirou, 2015-06-03:
Thank you! I will try it.

I have not read source code deeper, but if paypal/gatt does not work with `HCI_CHANNEL_RAW`, I think `newSocket()` should return err instead of trying HCI_CHANNEL_RAW.
https://github.com/paypal/gatt/blob/master/linux/device.go#L80 

And, since noble works well on the Edison, I hope paypal/gatt also work with old kernel. (I will read noble and bluez code later)

shirou, 2015-06-03:
I read noble code and found it invokes a small c program which uses bluez.
https://github.com/sandeepmistry/noble/blob/master/lib/linux/hci-ble.js#L7

I think this architecture is not suite for paypal/gatt, so I close this issue. 
I you have any other suggestion, I will appreciate.
Thank you.

closed 2015-06-03 by shirou

mrkm, 2015-08-04:
@shirou 

I want to use paypal/gatt on the Intel Edison `(Linux edison 3.10.17-poky-edison+ #1 SMP PREEMPT Fri Jun 19 12:06:40 CEST 2015 i686 GNU/Linux)`, for discovering BLE peripherals.

So I modified some files, and had succeeded it.
https://github.com/mrkm/gatt/commit/7734366089cd979468a5071e948a2fbf8bbf11ce

To use this patched codes, compile with `-tags=hci_channel_raw` option.
```
go build -a -tags=hci_channel_raw discoverer.go
```

(Make sure `rfkill unblock bluetooth` and `hciconfig hci0 up`)

shirou, 2015-08-04:
@mrkm wow, it's looks like great works!  I have not checked on my Edison yet, but tomorrow, I will do!

shirou, 2015-08-05:
@mrkm I confirmed your code works great on my Edison. Bravo!!
How about create new device struct and add to [HCI_CHANNEL_RAW handling](https://github.com/paypal/gatt/blob/master/linux/device.go#L80) ? Then, your code can be merged to this repository.

mrkm, 2015-08-06:
@shirou Thank you for your testing and kind advice. I'll give it a try!

abourget, 2015-11-13:
@mrkm any plans to rebase this and get it to be merged ?

mrkm, 2015-11-30:
@abourget Sorry, I don't have enough time to do that.

abourget, 2015-12-03:
I see there is some duplicate code.. is it a big effort to rework that patch into something elegantly mergeable ?  Do you think it takes a lot of familiarity with bluetooth's internals ?