paypal/gatt #43: Unhandled event

rabarar, 2015-06-22:
Not sure if this is an issue or the way the stack is supposed to work - but I get the following output and don't see how a callback would respond to it.

2015/06/22 10:39:37 Unhandled event: xpc.Dict{"kCBMsgId":53, "kCBMsgArgs":xpc.Dict{"kCBMsgArgDeviceUUID":xpc.UUID{0x4c, 0x33, 0x9e, 0x37, 0x99, 0x2a, 0x42, 0xb2, 0x9d, 0xaf, 0xf2, 0x8e, 0x3, 0xc6, 0xbf, 0x5e}, "kCBMsgArgATTMTU":104}}

mark2b, 2015-07-27:
Have same problem with discovered iOS device

flemay, 2015-08-04:
Same here with my Mac. I don't know where I can find the meaning of all the MsgId. Would anyone know? GATT doesn't handle the id 53 in function HandleXpcEvent (device_darwin.go).

However, the bluetooth indicator on my mac shows that it is connected to my board!

roylee17, 2015-08-04:
It's a MTU exchange request, which we currently don't handle on the OS X port.
In this case, the client (central) should fall back with the default MTU (23) or might drop the connection.

flemay, 2015-08-04:
@roylee17 Thanks! Would you know where I could find all the MTUs?

bgentry, 2015-08-28:
@roylee17 it looks like @flemay has made some progress on this on his fork. I have an iDevices iGrill2 device that never fully connects on paypal/gatt (halting at the unhandled event warning), but is able to fully discover services and characteristics with flemay/gatt. Would be nice to see that stuff merged in!

roylee17, 2015-08-28:
@bgentry That's sound nice. I looked at @flemay's branch, and it seems that some event IDs have been changed since Maverick.

@flemay, are you interested in sending a rebased PR for merging?

bgentry, 2015-08-28:
Yeah, I think looking at noble you'll see that they have different files
for Mavericks vs Yosemite support. I guess you'd need to switch on Darwin
version for the various event IDs.

bgentry, 2015-08-28:
I wonder if there's any OS X C constants we can match against that are more
stable across OS X versions.

flemay, 2015-08-29:
@bgentry @roylee17 My changes are mainly based on Noble. He told me the codes change whenever Apple releases a new os version. So he does reverse engineering to maintain Noble. I wish we could specify a version of os in golang the same way we target platform.

At the moment I am not 100% sure if the codes are exact. I am still trying to understand how I can subscribe to notify service using **p.SetNotifyValue**. Any clue?

roylee17, 2015-09-04:
Hi Guys,

Those event codes are reverse engineered by Noble/Bleno, and we're in debt to them :-)

Not sure what's the best way to tell the OS X version yet, but it seems the following should do the work, and we can replace it later when we find a better (canonical) way to do it.
```
        osver, err := syscall.Sysctl("kern.osrelease")
        if err != nil {
                return
        }
        fmt.Printf("%v\n", osver) // 14.5.0 on Yosemite
```

Are you interested in having those event codes merge back with support of Yosemite and Mavericks?
It's fine to me even it's not 100% exact. Once we have the support in place, other OS X users would be easier to find a place to keep it update.

Regarding p.SetNotifyValue(), the easiest way is to modify the examples/explorer.go and give it a try:
The  p.SetNotifyValue() has to be called after descriptors are discovered, or it returns "no cccd" error.

Roy
```
diff --git a/examples/explorer.go b/examples/explorer.go
index a25fa2f..ad2048a 100644
--- a/examples/explorer.go
+++ b/examples/explorer.go
@@ -8,6 +8,7 @@ import (
        "log"
        "os"
        "strings"
+       "time"

        "github.com/paypal/gatt"
        "github.com/paypal/gatt/examples/option"
@@ -116,9 +117,22 @@ func onPeriphConnected(p gatt.Peripheral, err error) {
                                }
                                fmt.Printf("    value         %x | %q\n", b, b)
                        }
+
+                       // Subscribe the characteristic, if possible.
+                       if (c.Properties() & (gatt.CharNotify | gatt.CharIndicate)) != 0 {
+                               f := func(c *gatt.Characteristic, b []byte, err error) {
+                                       fmt.Printf("notified: % X | %q\n", b, b)
+                               }
+                               if err := p.SetNotifyValue(c, f); err != nil {
+                                       fmt.Printf("Failed to subscribe characteristic, err: %s\n", err)
+                                       continue
+                               }
+                       }
+
                }
                fmt.Println()
        }
+       time.Sleep(5 * time.Second)
 }
```

flemay, 2015-09-27:
@roylee17 Thanks for the reply. I have tried your code but couldn't get it work on my `mac`. Following the [noble package](https://github.com/sandeepmistry/noble/blob/master/lib/mac/yosemite.js#L631), I made the SetNotifyValue to send `68` and then the response number to `74`. 

So It goes inside the `loop()` and the `rsp.id` is `74` but it does not have `kCBMsgArgIsNotification` field so it never executes the subscribed function. This happens one time.

The `loop()` will wait indefinitely. It feels like the `peripheral` never sends notifications.

roylee17, 2015-09-27:
So the above code subscribed to your peripheral's characteristic, but you're not sure does your peripheral actually send notification? 

You can test your peripheral with your phone (I use LightBlue on iPhone). Subscribe to the characteristic, and see if that peripheral does send some notification.

flemay, 2015-09-27:
It does work with `noble` and `lightblue` from my Mac without any problems!

roylee17, 2015-09-27:
Not sure this is due to the event ID changes between OS X version, or it's because didn't handle "Indication" correctly.

Since now I only have Yosemite, I'll see what's happening on Yosemite, and get back to you.

flemay, 2015-09-27:
It also works on my iphone with `lightblue`. 

I don't call `SetIndicateValue` in my code. I might have to push my latest modification of `flemay/gatt` so you can play with it!

flemay, 2015-09-27:
Pushed! :)

roylee17, 2015-09-28:
The rsp.id == 74 inside the loop() now should be changed to 71. It meant to tell weather a "read response" is a response to earlier read request, or is an asynchronous notification.

Now the problem is - there is no further report of event 71, even we seem to successfully subscribe to the characteristic. I can see my peripheral sending notifications after the subscription. But they are not reported. Will look into it.

flemay, 2015-09-28:
Thank you so much for investigating! :)

mattetti, 2015-10-10:
@flemay's MTU changes for OS X look good and are pretty much needed for anyone on Mac. I'm here because I'm hitting the same Yosemite bug others reported. Without a MTU update for Darwin (btw 53 seems to be missing), local dev is almost impossible :(

roylee17, 2015-10-11:
Sorry for the late response , and taking so long to resolve this (hopefully)

There are actually two issues here:

1. The device is not initialized correctly.
Though we did specify the MacDeviceRole for NewDevice(), I failed to use the specified value to initialize the device. In this case, the corebluetooth (OS X) doesn't report the "peripheralConnected:38" (and also characteristicRead :71) events. So the initialization code for the connection handling is not called.

9a3c51a7f30a07900479ab1b82a839968a4610b5 should fix this.

2. The event ID are changed.

The event IDs were ported from last year for Mavericks, so we need to update it for Yosemite.
Thanks for @flemay for porting the changes.

Let me know how it works for you :)

mattetti, 2015-11-23:
What's the status of this issue, is there anything I can do to help? (I'm on Mavericks and have an example code + BLE device)

flemay, 2015-11-23:
I will try the fix tonight and let you know! I don't know about Mavericks as I am working with yosemite/el-capitain :)

mattetti, 2015-12-01:
For what it's worth, I think it's totally fine to only target the latest 2 releases of OS X

barustnt, 2018-08-09:
hello
with go 1.10 and when i build server.go i have this message 
"undefined: central"
any help please !!?