paypal/gatt #44: OS X - server example

rabarar, 2015-06-22:
The discover and explorer examples seem to work on OS X, but the server example doesn't appear to work - I'm using the Apple Developer BLE PacketLogger app to monitor traffic and I don't see any advertisement packets being sent.

roylee17, 2015-06-22:
Did you mean you see the 'advertisement packet' happened once, or it was actually the SET_ENABLE_ADVERTISE command from the host to the device?

One thing can be confusing is that setting the device name is ignored by corebluetooth on OS X. So the device appeared on the LightBlue will be you computer name, instead of what you set in the API.

rabarar, 2015-06-22:
I was looking for the device name not my computer -- that must have been it. I'll go back and look...

rabarar, 2015-06-23:
Okay, I looked at the server example running both on linux and on OS X - things work as expected under linux, however, under OS X using LightBlue I see my computer name and No services listed. 

When I Select the peripheral to inspect it with LightBlue - the OS X Server prints out two unhanded events:

2015/06/23 09:46:52 Unhandled event: xpc.Dict{"kCBMsgId":53, "kCBMsgArgs":xpc.Dict{"kCBMsgArgDeviceUUID":xpc.UUID{0x4, 0xcd, 0xc1, 0x9f, 0xdd, 0xde, 0x44, 0x70, 0x86, 0x7f, 0x85, 0x64, 0xf5, 0xe8, 0x21, 0xe5}, "kCBMsgArgATTMTU":158}}
2015/06/23 09:46:52 Unhandled event: xpc.Dict{"kCBMsgArgs":xpc.Dict{"kCBMsgArgDeviceUUID":xpc.UUID{0x4, 0xcd, 0xc1, 0x9f, 0xdd, 0xde, 0x44, 0x70, 0x86, 0x7f, 0x85, 0x64, 0xf5, 0xe8, 0x21, 0xe5}, "kCBMsgArgATTMTU":158}, "kCBMsgId":53}
2015/06/23 09:46:52 Unhandled event: xpc.Dict{"kCBMsgId":82, "kCBMsgArgs":xpc.Dict{"kCBMsgArgConnectionInterval":15, "kCBMsgArgConnectionLatency":0, "kCBMsgArgSupervisionTimeout":200, "kCBMsgArgDeviceUUID":xpc.UUID{0x4, 0xcd, 0xc1, 0x9f, 0xdd, 0xde, 0x44, 0x70, 0x86, 0x7f, 0x85, 0x64, 0xf5, 0xe8, 0x21, 0xe5}}}

I see the UUID for the count service on LightBlue, but I don't understand how the UUID16 0x180F maps to the second UUID I see (or if that's even a UUID related to the server). 

Definitely seems more obfuscated on OS X than under linux.

achiku, 2015-06-23:
Hi, I ran into similar problem. The following is roughly what I did so far.

```
$ uname -a
Darwin akmac.local 14.3.0 Darwin Kernel Version 14.3.0: Mon Mar 23 11:59:05 PDT 2015; root:xnu-2782.20.48~5/RELEASE_X86_64 x86_64
```

```
$ mkdir sample && cd sample
$ go version
go version go1.4.2 darwin/amd64
$ go get github.com/mattn/gom
$ echo "gom 'github.com/paypal/gatt', :commit => 'fc9c77133e3b89c5ff904b9c7c41fee62f9ebab7'" > Gomfile
$ gom install
downloading github.com/paypal/gatt
$ cp _vendor/src/github.com/paypal/gatt/examples/server.go .
$ gom run server.go
```

Making sure it properly launch and I commented in the following line to debug.
https://github.com/paypal/gatt/blob/master/device_darwin.go#L348

```
$ rm -rf _vendor/pkg/darwin_amd64/github.com/paypal/*
$ gom run server.go
```

```
2015/06/23 23:00:42 >> 6, map[kCBMsgArgState:5]
State: PoweredOn
2015/06/23 23:00:42 >> 18, map[kCBMsgArgAttributeID:1 kCBMsgArgResult:0]
2015/06/23 23:00:42 >> 18, map[kCBMsgArgAttributeID:5 kCBMsgArgResult:27]
2015/06/23 23:00:42 >> 16, map[kCBMsgArgResult:0]
2015/06/23 23:00:42 >> 16, map[kCBMsgArgResult:0]
```

and then I use the following app to connect to server.
https://itunes.apple.com/us/app/lightblue-bluetooth-low-energy/id557428110?mt=8

The result is the log below.

```
2015/06/23 23:01:00 >> 53, map[kCBMsgArgDeviceUUID:3369fd6aa14d4d659af89c81305e686c kCBMsgArgATTMTU:158]
2015/06/23 23:01:00 Unhandled event: xpc.Dict{"kCBMsgId":53, "kCBMsgArgs":xpc.Dict{"kCBMsgArgATTMTU":158, "kCBMsgArgDeviceUUID":xpc.UUID{0x33, 0x69, 0xfd, 0x6a, 0xa1, 0x4d, 0x4d, 0x65, 0x9a, 0xf8, 0x9c, 0x81, 0x30, 0x5e, 0x68, 0x6c}}}
2015/06/23 23:01:00 >> 53, map[kCBMsgArgDeviceUUID:3369fd6aa14d4d659af89c81305e686c kCBMsgArgATTMTU:158]
2015/06/23 23:01:00 >> 53, map[kCBMsgArgDeviceUUID:3369fd6aa14d4d659af89c81305e686c kCBMsgArgATTMTU:158]
2015/06/23 23:01:01 >> 82, map[kCBMsgArgConnectionInterval:15 kCBMsgArgConnectionLatency:0 kCBMsgArgSupervisionTimeout:200 kCBMsgArgDeviceUUID:3369fd6aa14d4d659af89c81305e686c]
2015/06/23 23:01:01 Unhandled event: xpc.Dict{"kCBMsgArgs":xpc.Dict{"kCBMsgArgSupervisionTimeout":200, "kCBMsgArgDeviceUUID":xpc.UUID{0x33, 0x69, 0xfd, 0x6a, 0xa1, 0x4d, 0x4d, 0x65, 0x9a, 0xf8, 0x9c, 0x81, 0x30, 0x5e, 0x68, 0x6c}, "kCBMsgArgConnectionInterval":15, "kCBMsgArgConnectionLatency":0}, "kCBMsgId":82}
2015/06/23 23:01:01 >> 82, map[kCBMsgArgDeviceUUID:3369fd6aa14d4d659af89c81305e686c kCBMsgArgConnectionInterval:15 kCBMsgArgConnectionLatency:0 kCBMsgArgSupervisionTimeout:200]
```
It seems like ```kCBMsgId``` 53 is not defined in the switch statement and went stright to default clause. I googled around some time but couldn't find any list of ```kCBMsgId``` definition.

roylee17, 2015-07-15:
Looks like an MTU update request. We don't handle it write now.
Need more helpers to take of of the OS X implementation.

raff, 2016-02-04:
The OSX implementation was taken from my "port" of nodejs noble/bleno (see github.com/raff/goble) and I ported as much as I could of the original nodejs implementation at the time. Adding the processing of more messages shouldn't be very hard, if you know what they should be doing. If you can write down what the MTU update request should do I can probably add it.