paypal/gatt #51: didUpdateValueForCharacteristic() equivalent in GATT to read notification?

raowhizz, 2015-08-12:
Can someone tell me what is didUpdateValueForCharacteristic() equivalent in GATT ? Basically I want to read notification .

roylee17, 2015-08-12:
You can write your own handler, and register it with 
```
SetNotifyValue(c *Characteristic, f func(*Characteristic, []byte, error)) error
```

roylee17, 2015-09-10:
Regarding p.SetNotifyValue(), the easiest way is to modify the examples/explorer.go and give it a try.

Note:
1. The p.SetNotifyValue() has to be called after descriptors are discovered, or it returns "no cccd" error.
2. Add a sleep (delay) in the sample as well, in case the exploration finishes before the peripheral notifying some stuff. 

Roy
```
diff --git a/examples/explorer.go b/examples/explorer.go
index a25fa2f..ad2048a 100644
--- a/examples/explorer.go
+++ b/examples/explorer.go
@@ -8,6 +8,7 @@ import (
        "log"
        "os"
        "strings"
+       "time"

        "github.com/paypal/gatt"
        "github.com/paypal/gatt/examples/option"
@@ -116,9 +117,22 @@ func onPeriphConnected(p gatt.Peripheral, err error) {
                                }
                                fmt.Printf("    value         %x | %q\n", b, b)
                        }
+
+                       // Subscribe the characteristic, if possible.
+                       if (c.Properties() & (gatt.CharNotify | gatt.CharIndicate)) != 0 {
+                               f := func(c *gatt.Characteristic, b []byte, err error) {
+                                       fmt.Printf("notified: % X | %q\n", b, b)
+                               }
+                               if err := p.SetNotifyValue(c, f); err != nil {
+                                       fmt.Printf("Failed to subscribe characteristic, err: %s\n", err)
+                                       continue
+                               }
+                       }
+
                }
                fmt.Println()
        }
+       time.Sleep(5 * time.Second)
 }
```

closed 2015-09-10 by roylee17