paypal/gatt #57: Multiple connections

raowhizz, 2015-12-15:
As a central, how can I make multiple connections at the same time ? Also what is the max number of peripheral connections allowed by GATT  ?

roylee17, 2015-12-16:
Haven't tried multiple connections when running as a central, but did 4 concurrent connections when running as a peripheral. The number of concurrency is determined by the hardware/firmware.

moogle19, 2016-01-06:
@roylee17 Which OS and Hardware did you use to achieve 4 concurrent connections?

roylee17, 2016-01-06:
Broadcom - BCM4334

helinwang, 2016-03-23:
turns out you can do it having multiple connect on linux as a central. (do not know about other platforms)
first you have to do
```
d, err := gatt.NewDevice([]gatt.Option{
		gatt.LnxMaxConnections(5),
		gatt.LnxDeviceID(-1, true),
	}...)
```
Then, that is probably a bug, you can not connect to different devices at same time. Otherwise only first connected device gets the connected callback. You have to do the connection one at a time. I did it in following way:
```
mu.Lock()
go func(count int) {
	time.Sleep(time.Duration(count) * 5 * time.Second)
	p.Device().Connect(p)
}(count)
count++
mu.Unlock()
```