paypal/gatt #58: Can not scan peripheral device after running server for several minutes

shanwen0120, 2015-12-18:
I can scan and get the server after I build and run the sample server, but after running up the server about 20 minutes, i can not get the server anymore. (Scan by LightBlue in iPhone and hcitool in Linux)
If i shut down current unanswerable server and restart it, it appear immediately again.
Is there any hint about this problem?

Thanks

roylee17, 2015-12-18:
Which platform are you running on, Linux or OS X?
Do you need to access/connect to the server to trigger the problem, or just leave it advertising 20 mins and it happens? 

To debug on Linux platforms, you can hcidump the device which gatt is running on. It might be helpful to see what happened before it become unanswerable.

shanwen0120, 2015-12-18:
This problem happens on both platform, and i just scan the server without any access/connection after start it. 
I think hcidump is a good advice and i'll try it for figuring out the problem. 

Thank you, @roylee17

shanwen0120, 2015-12-22:
I can't get anything by hcidump when server disappear. The result of hcidump is below.
-------------------------------------------------------------------------------------------------
HCI sniffer - Bluetooth packet analyzer ver 5.28
device: hci0 snap_len: 1500 filter: 0xffffffff
 
 
< HCI Command: Read Local Supported Features (0x04|0x0003) plen 0
> HCI Event: Command Complete (0x0e) plen 12
    Read Local Supported Features (0x04|0x0003) ncmd 2
    status 0x00
    Features: 0xff 0xff 0xff 0xfe 0xdb 0xff 0x7b 0x87
< HCI Command: Read Local Version Information (0x04|0x0001) plen 0
> HCI Event: Command Complete (0x0e) plen 12
    Read Local Version Information (0x04|0x0001) ncmd 2
    status 0x00
    HCI Version: 4.0 (0x6) HCI Revision: 0x1e4d
    LMP Version: 4.0 (0x6) LMP Subversion: 0x3c55
    Manufacturer: Realtek Semiconductor Corporation (93)
< HCI Command: Read BD ADDR (0x04|0x0009) plen 0
> HCI Event: Command Complete (0x0e) plen 10
    Read BD ADDR (0x04|0x0009) ncmd 2
    status 0x00 bdaddr A0:0A:BF:15:27:70
< HCI Command: Read Buffer Size (0x04|0x0005) plen 0
> HCI Event: Command Complete (0x0e) plen 11
    Read Buffer Size (0x04|0x0005) ncmd 2
    status 0x00
    ACL MTU 1021:8 SCO MTU 255:16
< HCI Command: Read Class of Device (0x03|0x0023) plen 0
> HCI Event: Command Complete (0x0e) plen 7
    Read Class of Device (0x03|0x0023) ncmd 2
    status 0x00 class 0x000000
< HCI Command: Read Local Name (0x03|0x0014) plen 0
> HCI Event: Command Complete (0x0e) plen 252
    Read Local Name (0x03|0x0014) ncmd 2
    status 0x00 name 'RTK_BT_4.0'
< HCI Command: Read Voice Setting (0x03|0x0025) plen 0
> HCI Event: Command Complete (0x0e) plen 6
    Read Voice Setting (0x03|0x0025) ncmd 2
    status 0x00 voice setting 0x0060
< HCI Command: Set Event Filter (0x03|0x0005) plen 1
    type 0 condition 0
    Clear all filters
> HCI Event: Command Complete (0x0e) plen 4
    Set Event Filter (0x03|0x0005) ncmd 2
    status 0x00
< HCI Command: Write Connection Accept Timeout (0x03|0x0016) plen 2
    timeout 32000
> HCI Event: Command Complete (0x0e) plen 4
    Write Connection Accept Timeout (0x03|0x0016) ncmd 2
    status 0x00
< HCI Command: Read Page Scan Activity (0x03|0x001b) plen 0
> HCI Event: Command Complete (0x0e) plen 8
    Read Page Scan Activity (0x03|0x001b) ncmd 2
    status 0x00 interval 2048 window 18
< HCI Command: Read Page Scan Type (0x03|0x0046) plen 0
> HCI Event: Command Complete (0x0e) plen 5
    Read Page Scan Type (0x03|0x0046) ncmd 2
    0000: 00 00                                             ..
< HCI Command: LE Read Buffer Size (0x08|0x0002) plen 0
> HCI Event: Command Complete (0x0e) plen 7
    LE Read Buffer Size (0x08|0x0002) ncmd 2
    status 0x00 pktlen 0x001b maxpkt 0x10
< HCI Command: LE Read Local Supported Features (0x08|0x0003) plen 0
> HCI Event: Command Complete (0x0e) plen 12
    LE Read Local Supported Features (0x08|0x0003) ncmd 2
    status 0x00 features 0x0100000000000000 (Link Layer supports LE Encryption)
< HCI Command: LE Read Advertising Channel Tx Power (0x08|0x0007) plen 0
> HCI Event: Command Complete (0x0e) plen 5
    LE Read Advertising Channel Tx Power (0x08|0x0007) ncmd 2
    status 0x00 level 0x0 (dBm)
[... 274 lines cut ...]
  0000: 0c 0b 09 49 4e 46 41 4e  49 5f 42 4c 45 00 00 00  ...INFANI_BLE...
  0010: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Response Data (0x08|0x0009) ncmd 2
    status 0x00
< HCI Command: LE Set Advertise Enable (0x08|0x000a) plen 1
  0000: 00                                                .
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Advertise Enable (0x08|0x000a) ncmd 2
    status 0x00
< HCI Command: LE Set Advertising Data (0x08|0x0008) plen 32
  0000: 15 02 01 06 11 06 8d b9  7b e0 d5 b6 db b7 55 49  ........{.....UI
  0010: 3c c5 04 a7 c6 54 00 00  00 00 00 00 00 00 00 00  <....T..........
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Advertising Data (0x08|0x0008) ncmd 2
    status 0x00
< HCI Command: LE Set Advertise Enable (0x08|0x000a) plen 1
  0000: 01                                                .
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Advertise Enable (0x08|0x000a) ncmd 2
    status 0x00
-------------------------------------------------------------------------------
I check the code in device_linux.go is setting advertise mode as "LE General Discoverable Mode", so it shouldn't disappear after starting for a while, is there anyone can figure something out by the above result of hcidump?

Thanks so much.