paypal/gatt #64: OSX implementation will break with go 1.6

raff, 2016-02-04:
Go 1.6 has more restrictions in the use of cgo and Go pointers and the current implementation breaks in various places with various errors "panic: runtime error: cgo argument has Go pointer to Go pointer".

I have "fixed" my implementation with some casts (need to verify that it really works). See https://github.com/raff/goble/commit/811260b1d735999cdc026ab89dc513ad6658e29a

simonlinj, 2016-03-04:
I can confirm this issue.

Digitalblueeye, 2016-07-02:
It seems like there might be an error when trying to connect my OnePlus X to my Mac Mini with server.go (example). The phone never connects properly returning the "incorrect passkey" error regardless of the password typed.

tolleiv, 2016-08-15:
The changes made by @roylee17 solve this issue for me!