paypal/gatt #65: Failed to open device, err: EBADFD

dtony, 2016-02-19:
I am trying to run example/server_lnx on OpenWRT platform (mips32 linux) but it fail to launch with error :
`Failed to open device, err: EBADFD`

I am using this toolchain to compile : https://github.com/gomini/go-mips32.

I joined the strace output (with options -r -d) to help find the problem. 
I installed bluetooth related stuff by following this doc : https://wiki.openwrt.org/doc/howto/usb.bluetooth
and I disabled bluetoothd deamon. Also my Bluetooth dongle works well on my Linux amd64 box and is well recognize by `lsusb` on the OpenWRT device.

[strace_mips.txt](https://github.com/paypal/gatt/files/138364/strace_mips.txt)

roylee17, 2016-02-19:
```
0.003418 clock_gettime(CLOCK_MONOTONIC, {3559, 995447360}) = 0
0.002804 socket(PF_BLUETOOTH, SOCK_RAW, 1) = 4
0.002190 ioctl(4, HCIGETDEVLIST, 0x10464d98) = -1 EBADFD (File descriptor in bad state)
```
It failed at an very early stage to list/search available devices.
hciconfig also does that, so I'm wondering how did hciconfig work on your system?
The following is the relevant strace of hciconfig on an amd6x linux
```
socket(PF_BLUETOOTH, SOCK_RAW, BTPROTO_HCI) = 3
ioctl(3, HCIGETDEVLIST, 0x7f4cd184a010) = 0
ioctl(3, HCIGETDEVINFO, 0x7f4cd168fae0) = 0
```

Can you paste the strace of hciconfig on your mips box?

dtony, 2016-02-19:
`hciconfig` works fine, I am able to bring up hci0. Unfortunately I am away from the device until Monday so I would be able to send the strace output of hciconfig Monday. 

Thanks a lot.

dtony, 2016-02-22:
I made a strace of a working call to `hciconfig` and I don't really understand why it do not work.
Here is the relevant part of the trace for hciconfig : 
`
     0.007511 socket(PF_BLUETOOTH, SOCK_RAW, 1) = 3
     0.017718 brk(0)                    = 0x5fd000
     0.008581 brk(0x5fe000)             = 0x5fe000
     0.006768 ioctl(3, HCIGETDEVLIST, 0x5fd008) = 0
     0.007309 ioctl(3, HCIGETDEVINFO, 0x42ce14) = 0`

[hciconfig.strace.txt](https://github.com/paypal/gatt/files/140319/hciconfig.strace.txt)

brightshin, 2016-09-05:
How do you solve