paypal/gatt #75: Is this repo still being maintained?

jaredkwright, 2016-06-08:
It looks like the most recent commit was made eight months ago. Is there a better or more current fork I should be using?

eicca, 2016-08-26:
https://github.com/currantlabs/gatt looks like the most active fork. We've been able to use it successfully on a couple of projects.

roylee17, 2016-08-26:
https://github.com/currantlabs/gatt has served as a maintenance fork, and more importantly, a interim package during our transition to a new BLE package - https://github.com/currantlabs/ble 

We've been working on the **ble** for a while, and have ported all our internal projects from **gatt** to **ble**.  

The **ble** has been much more stable than *gatt* in terms of functionality, and the codebase is annotated wherever possible with the reference to the bluetooth spec. Though the interface / api is still evolving. It's always easy to modify the code to adapt the changes, compare to debug or fix issues in the legacy **gatt** repo.

So, checkout  the **ble**, and see if that serve your project better. Issues, and PRs are always welcome.

eicca, 2016-08-27:
Thanks for the explanation and the great work, @roylee17! We're going to check the **ble** project then.