paypal/gatt #76: Examples issue "Go pointer to Go pointer"

Digitalblueeye, 2016-06-26:
Last login: Sun Jun 26 18:24:04 on ttys000
Seths-Mac-mini:~ digitalblueeye$ go run /Users/digitalblueeye/Programming/Go/BLEServer.go 
panic: runtime error: cgo argument has Go pointer to Go pointer

goroutine 1 [running]:
panic(0x411ff20, 0xc82000a380)
	/usr/local/go/src/runtime/panic.go:481 +0x3e6
github.com/paypal/gatt/xpc._cgoCheckPointer0(0x40f2160, 0xc82000a360, 0xc82000a370, 0x1, 0x1, 0x4401cc0)
	??:0 +0x4d
github.com/paypal/gatt/xpc.XpcConnect(0x414d830, 0xf, 0x5804280, 0xc820086000, 0x0)
	/Users/digitalblueeye/Programming/Go/src/github.com/paypal/gatt/xpc/xpc_darwin.go:169 +0x1ba
github.com/paypal/gatt.NewDevice(0x41fe838, 0x1, 0x1, 0x0, 0x0, 0x0, 0x0)
	/Users/digitalblueeye/Programming/Go/src/github.com/paypal/gatt/device_darwin.go:67 +0x2a5
main.main()
	/Users/digitalblueeye/Programming/Go/BLEServer.go:13 +0x4f
exit status 2
Seths-Mac-mini:~ digitalblueeye$ 

using the latest version of Mac OS X on the mac mini (also latest). I get this issue when running explorer.go or server.go

jesselucas, 2016-07-01:
Worked around is to disable the cgo check `GODEBUG=cgocheck=0 `

Ex.
```
$  GODEBUG=cgocheck=0 ./server
```

kkdai, 2018-08-28:
In High Sierra it still another error  `Unhandled event: xpc.Dict{"kCBMsgId":4, "kCBMsgArgs":xpc.Dict{"kCBMsgArgState":5}}` which refer to https://github.com/currantlabs/ble/issues/58