paypal/gatt #80: Respond to Peripheral Connection Parameters Update Request

szyglowicz, 2016-09-08:
I am connecting to a Polar H7 chest strap.  If you look at the Polar Documentation, it says that if the connection is not setup correctly, after 30 seconds it will issue a Connection Parameters Update Request.  It does, but the GATT library just ignores it and prints "ignore l2cap signal" plus some data.  30 seconds later, it disconnects.  If I could get this to respond or set the parameters initially, then it would not disconnect.  Is there any way to make the connection parameters change per the peripherals request or set by default?

The Polar H7 wants:
Min Connection Interval: 312.25 ms
Max Connection Interval: 500 ms
Latency: 0
Timeout Multiplier 600 x 10ms = 6sec

I do similar process on Core iOS BTLE and the Polar never disconnects because I guess the core iOS library is handling this by itself.

Reference on how Polar H7 behaves: https://developer.polar.com/wiki/H6_and_H7_Heart_rate_sensors

roylee17, 2016-09-08:
Yet another reason to give the **ble** a try : https://gitub.com/currantlabs/ble.

We just had the same issues recently, and it was a quick and fix on the **ble**, but will need lots more work on paypal/gatt to make sure it works properly.

szyglowicz, 2016-09-08:
Ok, I will give it a try.  Thank you for your quick response.

szyglowicz, 2016-09-08:
https://github.com/currantlabs/ble is the correct link

roylee17, 2016-09-08:
Argh, right. Thanks!

szyglowicz, 2016-09-08:
Are there any other examples, notes or best practices associated with the library?

Steve

roylee17, 2016-09-09:
I pushed a next branch, which updates the API to take advantage of "context" for timeout and cancelations. 

Add a new example, blesh , which I think it might grow as a standalone utility later.
https://github.com/currantlabs/ble/tree/next/examples/blesh

szyglowicz, 2016-09-09:
I went to build blesh to test it but it missing a bunch of dependencies.  What is the best way around this besides start grabbing them one by one.

./../darwin/log.go:4:2: cannot find package "github.com/mgutz/logxi/v1" in any of:
	/usr/local/go/src/github.com/mgutz/logxi/v1 (from $GOROOT)
	/Users/steve/gogoband/code/go/src/github.com/mgutz/logxi/v1 (from $GOPATH)
../../gatt.go:8:2: cannot find package "github.com/pkg/errors" in any of:
	/usr/local/go/src/github.com/pkg/errors (from $GOROOT)
	/Users/steve/gogoband/code/go/src/github.com/pkg/errors (from $GOPATH)
../../darwin/adv.go:5:2: cannot find package "github.com/raff/goble/xpc" in any of:
	/usr/local/go/src/github.com/raff/goble/xpc (from $GOROOT)
	/Users/steve/gogoband/code/go/src/github.com/raff/goble/xpc (from $GOPATH)
filter.go:7:2: cannot find package "github.com/urfave/cli" in any of:
	/usr/local/go/src/github.com/urfave/cli (from $GOROOT)
	/Users/steve/gogoband/code/go/src/github.com/urfave/cli (from $GOPATH)

roylee17, 2016-09-09:
```
go get -u github.com/currantlabs/ble
cd ble
git checkout -b next -m origin/next
go get -u
```

`get -u github.com/currantlabs/ble` grabs the dependencies for the default branch - **master**. But I'm working on the **next** branch, which I'll merge it to **master** after some cleanup next week. In the mean time, check out the **next** branch manually, and `go get -u` again to bring the dependencies for the current branch.

szyglowicz, 2016-09-09:
Thanks for the details, I was figuring that out while you wrote it.  I still get the same results however when I go to build the original Explorer example.  It fails with:

# command-line-arguments
./main.go:48: undefined: ble.WithSighandler
./main.go:49: undefined: ble.Discover
./main.go:49: undefined: ble.FilterFunc

Is there something that changed?

roylee17, 2016-09-09:
Okay, just updated it. Give it another shot :)

szyglowicz, 2016-09-09:
Ok, I will.  I am running your shell on mac OS and it works fine to keep connected to Polar H7.  I have other devices I will try as well.  I am also running on raspberry Pi 3 and it is working well.  I no longer get the drops from the Polar H7 band, so it is better already.  I compile for Pi with:
GOOS=linux GOARCH=arm GOARM=7 go build -o blesh_lnx_arm7.o *.go

I will keep you up to date on how things are going on the Pi.  If I get some spare time, maybe I can help a bit on the project if you are interested.

roylee17, 2016-09-09:
Great! Looking forward to it.