paypal/gatt #81: Set up secure characteristics to trigger LE Secure connection

reallyroy, 2016-10-14:
Hi,

I noticed that the Characteristic struct has a secure field for secure properties. But it looks like the HandleRead and HandleWrite functions have secure property assignments commented out. When I uncomment the line, I don't even see the property on the client.

I was wondering if it would be possible to share an example on how to set up a secure write characteristic. The aim is to trigger a LE secure connection with and iOS application. The go code is the peripheral in my case.

Regards,
Roy

edillmann, 2017-01-02:
Hi,

I have the same problem, did you get any success on this ?

Regards,
Eric

bolteon, 2018-08-27:
Again, another instance of Adafruit not paying attention after months/years!