paypal/gatt #90: Failed to open device, err: no supported devices available

minhajakhterkhan199, 2017-06-03:
When I am running discoverer.go from example I am getting error no supported devices available. 
In addition, my Ubuntu 16.04 is successfully able to discover and connect to devices. 
Please advise hwo to solve the issue. 
[image]

go run examples/discoverer.go
2017/06/03 18:54:06 dev: hci0 up
2017/06/03 18:54:06 Failed to open device, err: no supported devices available
exit status 1

josephcenk, 2018-04-26:
For your kind info, already fixed that issue.

roylee17, 2018-04-26:
It takes root privilege and exclusive access to the HCI device.

1. Try execute the program with *sudo*. (but you mentioned that you already did.
2. See if there are other program grabbing the device, such as blued or other daemon.

BTW, more recent efforts for BLE with GO can be found at:

 https://github.com/currantlabs/ble
or
 https://github.com/go-ble/ble

Cheers

andrei-pavel, 2018-04-26:
I have recently found https://github.com/bettercap/gatt to be more active. *currantlabs* hasn't updated it years.

js-carlos, 2018-04-26:
I am using this fork https://github.com/cksmith/gatt haven't looked into bettercap

maheshjena1, 2019-03-26:
the issue still persists.
any idea on how to solve this?