paypal/gatt #91: Crashing on scan / connect loop (OSX)

matti777, 2017-06-16:
Hi, am trying to extend the Explorer example to print out the Manufacturer name Characteristic of the Primary Service of each device. However the library is quickly crashing to the following (or similar):

```
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x70 pc=0x43887c6]

goroutine 17 [running, locked to thread]:
github.com/paypal/gatt.(*device).HandleXpcEvent(0xc420076200, 0xc420236c90, 0x0, 0x0)
	/Users/matti/go/src/github.com/paypal/gatt/device_darwin.go:456 +0xe66
github.com/paypal/gatt/xpc.handleXpcEvent(0x4a00020, 0x4b002f0)
	/Users/matti/go/src/github.com/paypal/gatt/xpc/xpc_darwin.go:200 +0x209
github.com/paypal/gatt/xpc._cgoexpwrap_9c8af0d8d6e4_handleXpcEvent(0x4a00020, 0x4b002f0)
	github.com/paypal/gatt/xpc/_obj/_cgo_gotypes.go:458 +0x35

```

My main difference is Im not calling StopScanning() but instead trying to connect to each device as soon it is discovered and then extracting the manufacturer name. I would prefer not to connect but I assume it is mandatory for accessing the Characteristics? I am not familiar with BLE that well.

matti777, 2017-07-23:
Hmm so apparently this project is dead in the water. The lib seems highly unstable at least on OSX, so I am forced to look for alternatives. Anyone have nice suggestions? I plan to run the app on OSX, Desktop linux as well as Raspberry Pi so those platforms / architectures should be supported..

mikevoyt, 2017-07-24:
@matti777 take a look at this comment:
https://github.com/paypal/gatt/issues/75#issuecomment-242817481

The new project is here:  https://github.com/currantlabs/ble